meshtastic/Meshtastic-Android #280: device loosing bluetooth / ble pairing

hb9eue, 2021-05-25:
Hi

Samsung S8 Phone <=> tbeam with 1.2.30

When pairing with my Samsung S8, that pairing only is working for at max a couple of days. Then it requires re-pairing.

I think, it's not always after the same timespan. Sometimes re-pairing is required after a couple of hours, sometimes after a couple of days.

I think it more often happens, when the device is being operated on battery power (i don't know if low voltage might be the culpit, or the device entering low power mode)

I suspect, the key somehow is getting corrupted on the tbeam. I have another esp32 device paired with that S8 phone which is staying paired.

Have others also observed this issue? Is there a remedy?

-Benoit-

geeksville, 2021-05-25:
@hb9eue could you try this and report back in this bug (or in that forum thread?)

https://meshtastic.discourse.group/t/bluetooth-connection/152/15

geeksville, 2021-05-25:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/bluetooth-connection/152/16

hb9eue, 2021-05-25:
Hi

Thank you for the quick reply. I checked AVRCP settings. Samsung S8 ist set to 'default' 1.4. I'll try the other settings and see if the device can keep the paired status for longer.

closed 2021-12-16 by andrekir