meshtastic/Meshtastic-Android #281: No way to send a message without using the IME

jdstroy, 2021-05-28:
As it is [currently implemented](https://github.com/meshtastic/Meshtastic-Android/blob/0c7dac8cb2984352bb232e389b59588b22ebd028/app/src/main/java/com/geeksville/mesh/ui/MessagesFragment.kt#L231), there isn't a way to send a message without using the on-screen SIP/IME on Android.  Usually the soft input panel/IME on Android is disabled when a Bluetooth or integrated keyboard is attached; since the action of sending a message is bound to the Done IME event, users that aren't using the IME don't have a way to send messages.  Or for that matter, adjust other settings that are bound to the IME Done event (e.g. setting the owner name).

geeksville, 2021-05-28:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/1-2-31-feedback/3370/1

mc-hamster, 2021-11-18:
andrekir — Today at 11:05 AM
great!
https://github.com/meshtastic/Meshtastic-Android/issues/96 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/154 (date stamp OK)
https://github.com/meshtastic/Meshtastic-Android/issues/185 (Note sounds OK)
https://github.com/meshtastic/Meshtastic-Android/issues/194 (elapsed minutes OK)
https://github.com/meshtastic/Meshtastic-Android/issues/258 (Dark theme OK)
https://github.com/meshtastic/Meshtastic-Android/issues/279 (User reports resolved)
https://github.com/meshtastic/Meshtastic-Android/issues/111 (Solved in PR #234)
https://github.com/meshtastic/Meshtastic-Android/issues/281 (Solved on 1.2.45)
https://github.com/meshtastic/Meshtastic-Android/issues/289 (Solved on 1.2.45)

closed 2021-11-18 by mc-hamster