meshtastic/Meshtastic-Android #290: resend message on wake up fails in is_low_power mode

ChuckNorrison, 2021-08-16:
Testing the current pre release 1.2.43 alpha. Works so far except the power safe mode makes some trouble on send messages.

Testing with 3 TBEAM devices (v1.1 with current meshtastic fw 1.2.44). 
One device is set as is_router and two mobiles with is_low_power enabled.

iam connected via android app to one of the mobiles. I can send and receive messages from the mesh, while my device is connected. But if my low power tbeam goes into sleep and i try to send a message from android app, the message will stuck forever (cloud icon with up arrow is displayed).

I downgraded back to android app 1.2.31 beta and solved the behaviour for me.

KiwiHC16, 2021-09-03:
I m facing identical issue on 1.2.45.b674054 firmware and android 1.2.45

Sander85r, 2021-11-12:
I m facing exactly the same issue too on T-Beam v1.1 1.2.45 firmware and android 1.2.43 & 1.2.45

andrekir, 2021-12-30:
fix #330 available starting from [release 1.2.52](https://github.com/meshtastic/Meshtastic-Android/releases)

closed 2021-12-30 by andrekir