meshtastic/Meshtastic-Android #294: Duplication of messages

KiwiHC16, 2021-09-06:
I have 4 TBeam equipments all in the same room for tests.

All are with 5V to keep them awake.

One (c480) is connected to a linux system.
One (c500) is connected to Android App in BLE.

If I send a message from c480:
```
 meshtastic --sendtext "TYTY"
```
I get 2 messages TYTY on the android Application.
All TBeam print TYTY on their screen.

If I send uniquely to the c500 / Android:
```
meshtastic --sendtext "Tata" --dest \!f244c500
```
Then only the receiver print on the screen the message but the Android App show 2 messages.
In the debug page of the android App, I see only one message.

geeksville, 2021-09-06:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/android-app-1-2-45-ready-for-brave-alpha-testers/3793/5

andrekir, 2021-12-29:
@KiwiHC16 can you confirm if this issue is still present on newer versions of the app? I could not reproduce it on ver 1.2.51

closed 2022-01-09 by andrekir