meshtastic/Meshtastic-Android #298: QR Code not functioning Correctly

KrazyKerbalnaut, 2021-10-20:
Trying to use the QR code, it seems to bring the user back to the meshtastic.org site with an unknown address. Currently, it doesn't allow you to join a channel.

agw, 2021-11-10:
this is also happening for me on stock (pixel) android 12 and the latest android 12 build of GrapheneOS.
I also went into App info > Open by default and added "www.meshtastic.org" to the supported links on each device but am still unable to share the scanned QR code to the Meshtastic app, or open the link with anything other than my default web browser

KrazyKerbalnaut, 2021-11-15:
A work around I found is to use Google Lens' app. Kind of annoying to have to have an app just for it but it's all I got for now.

andrekir, 2021-12-19:
(ref #188) in-app QR code scanner added in version 1.2.51 plus detailed step-by-step on how to open and share channel links: https://meshtastic.org/docs/software/android/android-usage#join-a-channel

closed 2021-12-19 by andrekir