meshtastic/Meshtastic-Android #315: Documentation issue: Android might not connect www.meshtastic.org links to the app

prampec, 2021-12-01:
I have installed the app to my Android 12 phone and the channel links are opened within the browser, not by the Meshtastic app. The solution was to manually add "Open by default" link to the app.
See attached image.
[image]

prampec, 2021-12-01:
(Would be nice to provide an "enter connection info" option to be used instead of automatic link processing.)

andrekir, 2021-12-19:
(ref #188) in-app QR code scanner added in version 1.2.51 plus detailed step-by-step on how to open and share channel links: https://meshtastic.org/docs/software/android/android-usage#join-a-channel

closed 2021-12-19 by andrekir

prampec, 2021-12-19:
Sorry, but I cannot see this very case covered by the documentation.
(QR code scanner can be a great solution to override this problem. I will check.)

prampec, 2021-12-29:
Sorry didn't found the "Troubleshooting" pulldown. Now I believe the documentation covers this issue.