meshtastic/Meshtastic-Android #316: Relisting in google play - remaining tasks and people

geeksville, 2021-12-04:
Hi guys,

I was talking with @mc-hamster and it seems possible that @andrekir might be willing to become the new 'primary' meshtastic android dev.  Cool?  Also @mc-hamster said that he'd be willing to make the youtube video now needed when asking for background-location permissions from the google-play admins.

@andrekir what is your google friendly email address?  So I can add you as a collaborator on google-play?  Also I can email you the signing certs for the app/account.

Eventually I'd like to move the app into a true 'meshtastic' google-play account, but [apparently](https://support.google.com/googleplay/android-developer/answer/6230247?hl=en#zippy=%2Cstep-review-policy-guidelines-and-service-fee-tier%2Cstep-make-sure-your-google-play-developer-accounts-are-registered-and-active) that is most easily done only **after** we've reblessed it with google.  

I'm going to add the details on the reblessing process (from my emails with google) to this issue later today or tomorrow.

mc-hamster, 2021-12-04:
Ack :)

andrekir, 2021-12-05:
@geeksville appreciate the invite and would love to help. you can add: andrekir@pm.me

If you can also share the `google-services.json` used in the build, we're all set.

geeksville, 2021-12-06:
Super cool @andrekir.  I'm **so** glad you stepped up to lead android development!  I'll send you and mc an email shortly with the google-services.json and the android keystore for building/signing the APK.  Please treat these two files as **extremely** confidential.

I'm blessing both of you right now with google-play and I'm happy to help if/when you have questions.  I really hope I can be back helping (i.e. coding) with this effort soon.

geeksville, 2021-12-06:
and alas, I'm out of computer time for today.  I'll forward the google "new special location permissions needed in google-play" email thread tomorrow with some notes on next steps (essentially, just a youtube video and an email to google).

mc-hamster, 2021-12-06:
@geeksville what is needed to be in the video? I’ll solicit volunteers to start on the video so it’ll be ready by the time we are ready for review on the play store.

andrekir, 2022-02-04:
done!

closed 2022-02-04 by andrekir