meshtastic/Meshtastic-Android #334: App does not update location after connected to device

jaykayenn, 2021-12-30:
Seems the app doesn't update location of the currently connected device. It only gets location when the device is initially connected, but not after. Effectively, this means the app will not track locations of any node (except fixed nodes with preset location). 

- Start Meshtastic app and connect to a device
- If device already has GPS location, app will update with location (eg. fixed node preset)
- Else, no location is set in app
- After the app has already connected, GPS updates on device do not get updated to the app. 
- GPS coordinates can be seen updating on the device screen, but no change on app even after half an hour. 
- Disconnecting the device in app and reconnecting will update a new position from device to the app. But again, no further updates while connected. 

App: .52
Firmware: .50

andrekir, 2022-01-03:
can you confirm downgrading to device firmware 1.2.45 resolves this issue?

jaykayenn, 2022-01-03:
New finding:

- Pressing 'back' on the nodelist page causes the window to close and reveal another copy behind with the actual refreshed values. 
- Closing the app to homescreen and opening the app again will also (usually) cause the app to show refreshed values. Sometimes requires closing and opening a few times. 

Android 11