meshtastic/Meshtastic-Android #339: No GPS position is ever trasmitted to Heltec Lora 32 by Android app

mchisari, 2022-01-07:
**Description**
As described in (https://meshtastic.discourse.group/t/heltec-lora-32-issue/4414/4), the Android app doesn't seem to ever send GPS data to this device, so the owner position is never sent to mesh, despite the "provide location to the mesh" checkbox.

**To Reproduce**
- flashed device with device-install.sh firmware-heltec-v2.1-1.2.50.41dcfdd.bin
- Install android app
- pair via bluetooth
- enable "provide location to the mesh"
- give permission to access position (always) to Meshtastic app
- for 3 days went around house, using GPS with other apps without issues

**Expected behavior**
The location should be transmitted with other in node to mesh

**Screenshots**
No useful screenshots, I've pressed "file bug" button, don't know if/where data is sent...

**Device info:**
 - Device model: Heltec Lora 32 V2.1
 - Software Version: firmware-heltec-v2.1-1.2.50.41dcfdd.bin

**Smartphone information (if relevant):**
 - Device: Samsung M31
 - OS: Android 11
 - App Version 1.2.45

**Additional context**
I don't know if it's relevant, but I see no sign of position in "Debug Panel"
Android system tells that the app actually accessed GPS

Setting position via python's --setlon/setlat works fine.

mc-hamster, 2022-01-07:
moving to android project

geeksville, 2022-01-10:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/heltec-lora-32-issue/4414/16

closed 2022-01-19 by andrekir

micheljung, 2023-02-02:
Is this really fixed? 

I have two mobile phones, each connected to a Heltec LoRa 32 V2.0.
If I set a fixed location via the Meshtastic app, the position is sent and visible in both apps. If I remove the fixed location and want to get the location from the phone, however, it doesn't work on both phones.

The payload being sent for `POSITION_APP` is `%$\004\000\000`

andrekir, 2023-02-03:
1.2.45 is ancient and the code addressed here is long gone. now, when "provide location to mesh" is checked means location update is running, so definitely fixed. 

plain old LocationManager can be very slow (especially indoors and stationary), that's why I added back [FusedLocationClient for google flavor](https://github.com/meshtastic/Meshtastic-Android/commit/3c1629ffbdb0b94e5c042b04a0b36b525ff321b8) in 2.0.15. if you're running the google release, give it a shot.

micheljung, 2023-02-03:
Fantastic, I tested it and it works. Looks like this got released just in time :) thanks!