meshtastic/Meshtastic-Android #340: Export messages as CSV generate an empty file

ernax78, 2022-01-08:
When saving messages as CSV - it generates an empty file. Once it generated a file with headers only , but no messages.

[image]

Android 11,  App version 1.2.51

mcumings, 2022-02-03:
I started looking into this and found reference to the original author's use case here: https://meshtastic.discourse.group/t/save-position-messages-in-csv/2900

The original use case was to avoid the requirement to use the range test plugin to collect the data, as the plugin exposes the collected data via the embedded web server which is only available on ESP32- based devices.

All of that to say, the implementation only writes packets originating from remote nodes and ONLY if both the local node and the remote node are sharing location/position information. @ernax78 - Does this describe the scenario you are observing?

I believe the above makes sense for the original use case but maybe less so for a feature entitled generically as "Save messages as CSV..."

I propose that:
- We write ALL packets from remote nodes into the CSV.
- If distance can be calculated then that column will receive the calculated distance (in meters).
- If not enough information is present to compute the distance then we render a `?` into that column.

Given the columns being written ("From", "RSSI", "SNR", "Time", "Dist") I dont think it makes sense to write packets originating from the local host.

andrekir, 2022-02-03:
my suggestion is to use the same columns from the plugin generated `rangetest.csv` :

`"time, from, sender name, sender lat, sender long, rx lat, rx long, rx elevation, rx snr, distance, hop limit, payload"`

fill in what is available, render a `?` in what isn't.

mcumings, 2022-02-03:
I like your idea better. 😄   I've seen casual chatter on discord about the potential for freeing up space on the device by removing HTTP support, since it is both large and only on a subset of devices.  If that becomes a serious direction and comes to pass then the ability to get the entire dataset via the app - versus a subset - becomes more necessary.

Also, thinking more about it I think CSV files conventionally use empty fields to indicate no data available so I'm inclined to drop the `?` as well.  i.e., If we have the data then we render the field data followed by the delimiter, otherwise we just render the delimiter.

mcumings, 2022-02-03:
Cross reference to the range test plugin source: https://github.com/meshtastic/Meshtastic-device/blob/master/src/plugins/esp32/RangeTestPlugin.cpp#L257

PR #365 (2022-02-04): Fixes #340 Improve CSV file export
Fixes  #340 and expands upon the written fields such that it (mostly) matches the range test plugin's exported file content.

Restructured this a bit to get more business logic out of the activity layer.  I moved it down to the viewmodel for now but this would really be moved into a repository.
commit 6012bddb (2022-02-04): Fixes #340 Improve CSV file export
commit a0d00a42 (2022-02-04): Merge pull request #365 from mcumings/fix-340

Fixes #340 Improve CSV file export
closed 2022-02-04 by andrekir