meshtastic/Meshtastic-Android #344: Feature Request: Ability to message outside primary channel

VladTheSkewerer, 2022-01-13:
Didn't see this explicitly listed, but it's probably on a roadmap somewhere.

Can we get a method for selecting a channel other than the primary for messaging?  It would also be great to have a way to make it sticky so that channel is selected when the app is restarted, though that's just secondary.

Thanks for all your work!

cyberlink1, 2022-01-13:
This would explain why I cant seem to find a way to send a message on a given channel from the android ap. lol

Ill second this one, I have several channels set up on my units

charminULTRA, 2022-05-16:
Would also like to see this feature

phazed0, 2022-07-05:
I would also like to see this. The ability to have a Public, Private and Admin Channels accessible from the app would be killer. It would be really nice to perhaps have a --ch-add functionality with --ch-index and PSK options at time of channel creation(in the app). All in good time, thanks for everyone's hard work on this project!

garthvh, 2022-07-05:
moving this to device where this would be implemented