meshtastic/Meshtastic-Android #359: add offline map options to mapbox

andrekir, 2022-02-01:
Mapview
---

- [ ] migrate to mapbox API v10 ([Installation](https://docs.mapbox.com/android/maps/guides/install/), [Migrate to v10](https://docs.mapbox.com/android/maps/guides/migrate-to-v10/))
- [ ] add option to import map tiles / offline maps

garthvh, 2022-02-01:
@joshpirihi there is some interest in doing the same thing you did for the iOS app in android app, can you help @andrekir with any tile questions he runs into?  @thebentern has volunteered to do some android testing.

QGIS is the application we have been using to make the maps, the built in OpenStreetMap layers are a good place to start.

joshpirihi, 2022-02-01:
Happy to help! From what I could tell, MapBox doesn't have native support for local mbtiles map files. 

But I have seen a hack where you run a local http server on some random port (on the device), and tell the mapbox sdk that is a network source of mbtiles data.

PWRxPSYCHO, 2022-02-05:
> ## Mapview
> 
>     * [ ]  migrate to mapbox API v10 ([Installation](https://docs.mapbox.com/android/maps/guides/install/), [Migrate to v10](https://docs.mapbox.com/android/maps/guides/migrate-to-v10/))
> 
>     * [ ]  add option to import map tiles / offline maps

I've gotten some of the map migrated over to V10.

Still need to fix the following:
 - Gesture Controls
 - Adding Image,Source & layers
 - Updating Nodes
 - Camera Movement
 - Remove old dependency
 
 
I can open a PR with the changes so far.

andrekir, 2022-02-05:
> I can open a PR with the changes so far.

@PWRxPSYCHO great! please do.

PWRxPSYCHO, 2022-02-10:
@joshpirihi I'll have to look into .mbtiles. I'm not sure what the properties are on that file and if we can convert them over to what the offline manager needs to load the map (I would assume it is possible).

[MapBox V10 Offline Android SDK](https://docs.mapbox.com/android/maps/guides/offline/)

PWRxPSYCHO, 2022-02-10:
I will investigate this option and see what I can do! It looks like MapBox has an example one I can mess around with which is perfect

PWRxPSYCHO, 2022-02-21:
I've been chugging along at this and I think I'm about 75% of the way through getting something working. (There are a few menus/quality of life improvements I need to make still).

But I was reading into .mbtiles and I found that no mobile SDK can use them directly. [MBTiles](https://docs.mapbox.com/help/glossary/mbtiles/). 

Directly from that link:
**"_Note that the mobile SDKs do not support offline use of MBTiles directly. MBTiles can be used offline if they are part of a published Mapbox style._"**

I took this into consideration when trying to design the prototype offline-capability. A user would have to use a URI from their custom MapBox Style in order to download a specific area/location/overlay/etc in order to use MBTiles. Otherwise, they will be able to select a point on the map and download that region directly with their device.

For example: this Mapbox Style (mapbox://styles/pwrxpsycho/ckzrtyx12000514quk5ehz1tw) incorporates the sample [trails.mbtiles](https://docs.mapbox.com/help/data/trails.mbtiles) into its style which can then be downloaded onto the app and loaded as an offline region. 

[Preview Link of sample style](https://api.mapbox.com/styles/v1/pwrxpsycho/ckzrtyx12000514quk5ehz1tw.html?title=view&access_token=pk.eyJ1IjoicHdyeHBzeWNobyIsImEiOiJja3l3YWRteTgwNmpoMm5vMjU2Zzlxc2g5In0.SpidqqQbISBBnzAneyVPtA&zoomwheel=true&fresh=true#14/37.2686/-112.9425)

joshpirihi, 2022-02-21:
This sort of thing makes me really despise Mapbox! It seems a little bit ironic that the only way to have an offline map is to first upload it to the internet.

I did find [this](https://gist.github.com/typebrook/7d25be326f0e9afd58e0bbc333d2a175) which is a bodge that might improve offline mbtiles support.  It is a shame that Mapbox don't have first class offline support for their own file format (MapLibre, the open source fork of Mapbox SDK, [does support offline mbtiles as a native source ](https://github.com/maplibre/maplibre-gl-native/issues/17))

That said, thanks for all the good work you're doing!  I do appreciate it and don't want to direct any ill feelings towards the meshtastic dev team here, you guys are awesome!  I realise we're a bit stuck with Mapbox for now and I don't have the time to help with this issue.

joshpirihi, 2022-02-21:
Then again, I am that weirdo with my own map that I drew up.  Your current approach will work well for most users, which is good

garthvh, 2022-02-21:
Do we know what are our current MapBox api limits are?  I saw 15,000 on an old thread but that seems really low (i get 250,000 in TestFlight on iOS, and essentially unlimited in the App Store for my $99 a year developer account). Looks like the mapbox offline api is .25 cents per 1000 tiles if you use the ability to draw the region.

closed 2022-03-12 by andrekir