meshtastic/Meshtastic-Android #381: Device sleeping, but rather lost BT connection

KrustyDerClown, 2022-02-22:
Device T-Beam 1.1 Software 1.2.52
Huawei P30, Android 10 , app 1.2.52 

Device is paired with App.
Issue: After some time not using the app, or running in the background the device is shown as "sleeping"  At the top the cloud shows an arrow upwards and now message can be send, nor any information can be read in debug panel. The device itself seems to work normal. This can be fixed by simple dis- and enable bluetooth that instantly shows the app as connected (cloud with checkmark) 
Maybe some android sleeping functionaltity or bug?

caveman99, 2022-02-22:
Please update your device and app to the latest 1.2.x alpha available, there have been lots of bug fixes in the sleep code since 1.2.52. For the app you need to sideload it, the version on the store is too old.

KrustyDerClown, 2022-02-22:
I will try this...

KrustyDerClown, 2022-02-23:
Updated app and device to latest alpha, and also disable "Huawei" setting to automatic handle the powermanagement. Now the device is shown as sleeping for a short moment and then it changes to connected after a short period of time (1-3 seconds). Unsure if the PM setting did also influences the effect. Issue can be closed..

closed 2022-02-23 by andrekir