meshtastic/Meshtastic-Android #422: Feature Request: Allow App to be Exited

tomcarlson, 2022-04-26:
I personally do not like that it is impossible for me to stop the Meshtastic App without using Android to  "Force Stop" it.  It probably is not hurting anything for it to be always running, but I have enough OCD that it bugs me not be be able to make the app actually quit running.

Orbot (Tor App) is an example of an app that likes to stay running in the background, but it has a menu option that allows you to exit it.  I wonder if the same scheme could be added to Meshtastic.

mc-hamster, 2022-04-26:
Closed.

Duplicate of: https://github.com/meshtastic/Meshtastic-Android/issues/186

closed 2022-04-26 by mc-hamster

andrekir, 2022-04-26:
@tomcarlson the app can be gracefully closed when no interfaces are open, ie. "None (disable)" selected on the Settings tab.