meshtastic/Meshtastic-Android #425: Feature request: Implement ability to scan channel settings from image

DarkFIxED, 2022-05-03:
As a user I want to be able to share generated QR code with my teammates and he are able to open it via Meshtastic app. Currently I should scan it via another device through screen.

andrekir, 2022-05-03:
you can use the camera ("scan" button) from the app or the channel URL (ie.: https://www.meshtastic.org/d/#CgUYAyIBAQ) from anywhere else to join channels, check: https://meshtastic.org/docs/software/android/android-usage#join-a-channel

andrekir, 2022-05-03:
please search issues to verify the bug or feature has not been raised before. if you have questions join us at https://meshtastic.discourse.group or Discord: https://discord.gg/ktMAKGBnBs

closed 2022-05-03 by andrekir