meshtastic/Meshtastic-Android #442: Android 8 - the plus button doesn't work

megabyte2010, 2022-06-09:
The plus button to find new bluetooth device doesn't find anything. In the older versions without that button, I can pair without problems. Huawei p 10 lite.

JeffRockatansky, 2022-08-16:
Have you tried to turn on the location access and allow it to the Meshtastic app?

megabyte2010, 2022-08-16:
> Have you tried to turn on the location access and allow it to the Meshtastic app?

Yes, all permissions have been granted. It works in android 9 and 12 but not in android 8.

andrekir, 2022-08-16:
changing title for clarity since it is not an Android 8 issue. some phones have buggy OS implementation for companion device pairing service and don't work. not an app issue we could fix.

leaving this open to track affected devices.

megabyte2010, 2022-08-16:
> changing title for clarity since it is not an Android 8 issue. some phones have buggy OS implementation for companion device pairing service and don't work. not an app issue we could fix.
> 
> leaving this open to track affected devices.
The older meshtastic android versions that doesn't have that button can find the meshtastic device in this phone.

mcayea95, 2022-12-12:
Phone: Galaxy A53 running stock Android 13
Meshtastic App: 2.0.6
Tbeam Firmware: 2.0.6.97

Same issue.

andrekir, 2023-04-07:
since release 2.0.7 phones with Bluetooth discovery issues can pair to radios via Android Settings and they will show in the App's Settings tab.

closed 2023-04-07 by andrekir