meshtastic/Meshtastic-Android #453: USB cable connection instead of BLE

boby1975, 2022-06-23:
Hi,
Is it possible to use permanent USB cable connection instead of BLE connection with Meshtastic device like T-ECHO with USB port?
Thnks.

andrekir, 2022-06-24:
yes, see: https://github.com/meshtastic/Meshtastic-Android/blob/master/app/src/main/java/com/geeksville/mesh/repository/usb/README.md

if you have questions please post at https://meshtastic.discourse.group or join our Discord https://discord.gg/TYHeJpPc

boby1975, 2022-06-24:
Thanks a lot.
I've done this but there is no connection to mesh network using USB cable. There is no contacts.
With BLE all is ok.

Debug info
`D/com.geeksville.mesh.repository.usb.UsbBroadcastReceiver: USB device '/dev/bus/usb/001/007' was attached
D/com.geeksville.mesh.MainActivity: langCode is zz
D/com.geeksville.mesh.model.UIViewModel: ViewModel created
I/DecorView: [INFO] isPopOver=false, config=true
I/DecorView: updateCaptionType >> DecorView@45b1b09[], isFloating=false, isApplication=true, hasWindowDecorCaption=false, hasWindowControllerCallback=true
D/DecorView: setCaptionType = 0, this = DecorView@45b1b09[]
D/com.geeksville.mesh.MainActivity: Handle USB device attached! UsbDevice[mName=/dev/bus/usb/001/007,mVendorId=9114,mProductId=17413,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=TTGO,mProductName=TTGO_eink,mVersion=1.00,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@5853d27, mHasAudioPlayback=false, mHasAudioCapture=false, mHasMidi=false, mHasVideoCapture=false, mHasVideoPlayback=false, mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=160,mMaxPower=50,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=TinyUSB Serial,mClass=2,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=8,mInterval=16]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=64,mInterval=0]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=64,mInterval=0]]]]
D/com.geeksville.mesh.MainActivity: We have our required permissions
D/com.geeksville.mesh.MainActivity: Binding to mesh service!
I/com.geeksville.mesh.service.MeshService$Companion: Trying to start service debug=true
D/InputTransport: Input channel constructed: '1a465dd', fd=178
I/ViewRootImpl@3a1be17[MainActivity]: setView = com.android.internal.policy.DecorView@45b1b09 TM=true
I/com.geeksville.mesh.service.MeshService: Requesting foreground service=true
I/SurfaceControl: assignNativeObject: nativeObject = 0 Surface(name=null)/@0x6b1fd46 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1836 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
I/ViewRootImpl@3a1be17[MainActivity]: Relayout returned: old=(0,0,720,1600) new=(0,0,720,1600) req=(720,1600)0 dur=16 res=0x7 s={true -5476376666793172992} ch=true fn=-1
I/ViewRootImpl@6f90ef9[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
D/com.geeksville.mesh.MainActivity$mesh$1: Switching to USB radio /dev/bus/usb/001/007
D/com.geeksville.mesh.service.MeshService: Passing through device change to radio service: ...007
I/com.geeksville.mesh.repository.radio.RadioInterfaceService: stopping interface com.geeksville.mesh.repository.radio.SerialInterface@94529fc
D/com.geeksville.mesh.repository.radio.SerialInterface: Closing stream for good
D/com.geeksville.mesh.repository.usb.SerialConnectionImpl: Waiting for USB manager to stop...
D/com.geeksville.mesh.repository.radio.RadioInterfaceService: Setting bonded device to ...007
I/com.geeksville.mesh.repository.radio.RadioInterfaceService: Starting radio ...007
I/com.geeksville.mesh.repository.radio.SerialInterface: Opening com.hoho.android.usbserial.driver.CdcAcmSerialDriver@5f3e6da
D/CdcAcmSerialDriver: trying default interface logic
D/CdcAcmSerialDriver: claiming interfaces, count=2
D/CdcAcmSerialDriver: Control iface=UsbInterface[mId=0,mAlternateSetting=0,mName=TinyUSB Serial,mClass=2,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=8,mInterval=16]]
D/CdcAcmSerialDriver: data iface=UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=64,mInterval=0]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=64,mInterval=0]]
D/UsbRequestJNI: init
D/com.geeksville.mesh.repository.usb.SerialConnectionImpl: Starting serial reader thread
D/com.geeksville.mesh.repository.usb.SerialConnectionImpl: writing 4 byte(s)
I/SerialInputOutputManager: Running ...
D/com.geeksville.mesh.repository.radio.RadioInterfaceService: Broadcasting connection=true
D/com.geeksville.mesh.service.MeshService: Discarding NodeDB
D/com.geeksville.mesh.MainActivity$mesh$1: Service provided 101 messages and myNodeNum null`

`D/com.geeksville.mesh.ui.ContactsFragment: New contacts received: 2
I/com.geeksville.mesh.service.MeshService: in connectionState=DISCONNECTED
I/com.geeksville.mesh.service.MeshService: in getOnline, count=0
E/toRemoteExceptions: Uncaught exception, returning to remote client
W/com.geeksville.mesh.model.UIViewModel: Ignoring failure to get myId, service is probably just uninited... We don't yet have our myNodeInfo
D/com.geeksville.mesh.MainActivity: connchange DISCONNECTED -> DISCONNECTED
D/com.geeksville.mesh.MainActivity$mesh$1: connected to mesh service, isConnected=DISCONNECTED`

`D/CommonUsbSerialPort: Wrote 4/4 offset 0/4 timeout 0
I/SurfaceControl: nativeRelease nativeObject s[-5476376667011584864]
[... cut ...] lbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
I/ViewRootImpl@6f90ef9[MainActivity]: Relayout returned: old=(0,0,720,1600) new=(0,0,720,1600) req=(720,1600)8 dur=14 res=0x5 s={false 0} ch=true fn=1920
D/CommonUsbSerialPort: Wrote 7/7 offset 0/7 timeout 0
I/ViewRootImpl@6f90ef9[MainActivity]: stopped(true) old=false
D/com.geeksville.mesh.MainActivity: Unbinding from mesh service!`

closed 2022-07-29 by andrekir