meshtastic/Meshtastic-Android #459: Error: Code scanner module is not supported on current Google Play Services version, please upgrade

mc-hamster, 2022-08-10:
I have 1.3.30 installed on an LG L322DL running Android 10 and when I try to scan a QR code, I get the error message:

"Code scanner module is not supported on current Google Play Services version, please upgrade"

I clicked the Report Bug button at 10:30pm.

Douile, 2022-08-11:
QR scanner doesn't work at all on phones without google play services, maybe its worth swapping to a different QR scanning library.

andrekir, 2022-08-14:
@mc-hamster thanks for the report. fix is incoming.

@Douile thanks for mentioning aosp/microg. the app already uses zxing and handles channel url intents. play services code scanner is only meant for play store installs, so that needs fixing.

Douile, 2022-08-14:
Oh mb i should have read the code instead of assuming. Sorry for off topic but is there a build flag or something i need to turn on to use zxing instead of play services?

andrekir, 2022-08-14:
#461 should fix it. if you want to try the debug apk from that PR you can find it [here](https://github.com/meshtastic/Meshtastic-Android/actions/runs/2857532086).

andrekir, 2022-08-16:
@mc-hamster when you have the time, please confirm if [1.3.38 release](https://github.com/meshtastic/Meshtastic-Android/releases/tag/1.3.38) fixes this issue for you.

closed 2022-08-25 by andrekir