meshtastic/Meshtastic-Android #463: QR Scan on De-Googled phones not working

JeffRockatansky, 2022-08-16:
# System info

* LineageOS: 18.1 (2022-08-17)
* Android: 11
* Phone: OnePlus 8 Pro
* GApps: microG core
* Meshtastic app: 1.3.30
* LORA Device: T-Echo
* Meshtastic LORA Firmware: 1.3.30

# Description

When using an De-Googled Android phone, on unlocked channel options tab the "Scan" button doesn't work and throws an error instead.

# Expected behavior

1. Go to "channel options" tab
2. Push lock button to unlock options
3. Push "Scan" button, which opens a scanner camera
4. Scan channel info from QR code
5. Communicate through a new channel

# Actual behavior

1. Go to "channel options" tab
2. Push lock button to unlock options
3. Push "Scan" button
4. Get a popup with an error message: "Code scanner module is not supported on current Google Play Services version, please upgrade."

# Suggested fixes

1. Remove "Google Play Services Code scanner module" dependency from Meshtastic app
2. Register the app as a Meshtastic URL handler
3. Create and handle a custom URL Scheme (e.g. `mesh://`)

garthvh, 2022-08-16:
Pull requests are welcome, seems like partnering with the user who has been working on a fdroid build for this is the way to go if you want these features https://mesh.tastic.app/fdroid/repo/ as de-googled phones are optomistically one percent of the android user base.

andrekir, 2022-08-16:
Duplicate of #459

closed 2022-08-16 by andrekir