meshtastic/Meshtastic-Android #464: migrate messages to Room database

andrekir, 2022-08-16:
* [ ] 1. create Room database for Messages
* [ ] 2. refactor MeshService to save message DataPackets to new database
* [ ] 3. refactor viewmodel/MessagesFragment pointing to new database
* [ ] 4. remove message storage from MeshService

andrekir, 2022-08-16:
Entity draft:

data class Message(
    @PrimaryKey(autoGenerate = true) val uuid: Long,
    @ColumnInfo(name="contact") val contact: String,
    @ColumnInfo(name="channel") val channel: Int,
    @ColumnInfo(name="received_date") val received_date: Long,
    @ColumnInfo(name="datapacket") val datapacket: String)

- 'contact' filters direct messages to/from contacts
- 'channel' filters different broadcast channels
- 'received_date' sorts by received (new/old first)
- 'datapacket' is the raw DataPacket

Douile, 2022-08-17:
A few observations:

Would it be worth storing the message and status as well so that decoding the datapacket isn't necessary when rendering messages? or is this always going to be decoded anyway (e.g. in repository).

As the packets are already stored in a database it seems unnecessary to have the raw data packet for each message maybe we could just have a reference (foreign key) to the row in the packet table. Although I'm not sure if you mean

> 4. remove message storage from MeshService

to stop storing message packets in that database.

andrekir, 2022-08-17:
> 4. remove message storage from MeshService

for that I mean getting rid of [recentDataPackets](https://github.com/meshtastic/Meshtastic-Android/blob/master/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L565).

> Would it be worth storing the message and status as well so that decoding the datapacket isn't necessary when rendering messages? or is this always going to be decoded anyway (e.g. in repository).

good point. `status`, yes. it is probably the only field that gets modified/updated, others are static. `message`, I'm not sure how much of a performance hit it takes to decode the DataPacket on-the-fly. I believe we would also need other fields like `to` and `from`.

> As the packets are already stored in a database it seems unnecessary to have the raw data packet for each message maybe we could just have a reference (foreign key) to the row in the packet table.

PacketDao use in the Debug Panel can be confusing, I wonder if renaming to something else like PacketLogDao, DebugLogDao or MeshLogDao would be more appropriate. for instance, the option to "delete all" from the debug panel could cause problems with this reference, but that would be easy to remove.

note: I edited the entity draft above with some changes:
- renamed Message to DataPacket;
- added portnum (datapackets are not only used for messages);
- added status / messageStatus enum;
- corrected the 'datapacket' field to use DataPacket structure (not String).
- changed `received_date` to `received_time`

Douile, 2022-08-17:
aha yes I was confused, thought because it was Message, was exclusively TEXTAPP.

I doubt there will be much performance loss from decoding on the fly (completely based on assumption), but storing like that does prevent the need to update table structure each time the protobufs change.

I see what you mean with the delete all, with packet log also being limited to 500 packets that would also create dependency issues.

andrekir, 2022-08-17:
for some context: messages use a protobuf [MeshPacket](https://github.com/meshtastic/Meshtastic-protobufs/blob/master/mesh.proto#L646) structure.

MeshService converts MeshPackets to DataPackets for storage/usage. main difference is adding status field plus to/from are converted from nodeNum (int) to nodeID (string based on device mac address).

it would be more efficient overall to eliminate DataPacket and use straight MeshPacket, but that would add complexity to this refactoring and it can be done later.

Douile, 2022-08-26:
Is anybody working on this and are we happy with the design?

andrekir, 2022-08-26:
@Douile pushed my draft for the database: https://github.com/meshtastic/Meshtastic-Android/tree/meshpacket

some notes:
- renamed it MeshPacket because two classes named DataPacket was confusing;
- still undecided if tagging DataPacket or MeshPacket type [here](https://github.com/meshtastic/Meshtastic-Android/blob/meshpacket/app/src/main/java/com/geeksville/mesh/database/entity/MeshPacket.kt#L18) (I made a Converter for both);
- this last iteration I tried adding [DataPacket fields](https://github.com/meshtastic/Meshtastic-Android/blob/meshpacket/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L568) `from_id` and `to_id` (both converted by nodes `nodeNums` [here](https://github.com/meshtastic/Meshtastic-Android/blob/meshpacket/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L398)) (being able filter contacts using `from_id` and `to_id` would have to be investigated for this to work).

the idea was to have all data from DataPacket & MeshPacket in the DB, but just using DataPacket type as drafted in this issue will work and be more simple. feel free to use the code however you prefer.

closed 2022-09-16 by andrekir