meshtastic/Meshtastic-Android #471: Containts tracking libraries

jgoerzen, 2022-08-25:
TC on Android reports that Meshtastic contains:

- Google AdMod
- Google CrashLytice
- Google Firebase Analytics
- Mapbox

Even though I unchecked the box to send anonymous things, TC still blocked Meshtastic attempting to contact firebase-settings.crashlytics.com and firebaseinstallations.googleapis.com, as well as events.mapbox.com.

For an application that is focused on privacy and security, isn't this at cross-purposes with what we should be doing?

andrekir, 2022-08-25:
https://meshtastic.org/docs/legal/privacy

Meshtastic is free open source software made by volunteers, if there is anything wrong with our code contributions are always welcome. there are many reasons behind our library choices, #6 is a good start. also check #470.

closed 2022-08-25 by andrekir

jgoerzen, 2022-08-26:
Thank you.  That's helpful context.  I don't see anything about why AdMob is there though, or why things are sent even when the box is unchecked.  If you have context on those, that would be great!  Thanks!

garthvh, 2022-08-27:
> Thank you. That's helpful context. I don't see anything about why AdMob is there though, or why things are sent even when the box is unchecked. If you have context on those, that would be great! Thanks!

Google owns all those things other than mapbox, I don't think the Meshtastic app is using admob functionality, but I am pretty sure AdMob is included in the firebase SDK.