meshtastic/Meshtastic-Android #478: Send projected, labelled marker by tapping in map

MaximKrasikov, 2022-09-08:
would it be possible to implement sending a location on the map, as a named marker displayed on the map, without having a device at that specific location? It would be great to be able to e.g define a common meeting point that appears on everyone’s map without having to be at that point already.

andrekir, 2022-09-08:
this can be done adding support for Waypoint messages: https://meshtastic.org/docs/developers/protobufs/api#waypoint

MaximKrasikov, 2022-09-08:
will i be able to send list of markers to other users via waypoint ? Do I need to fix the firmware of the device and proto for this?

andrekir, 2022-09-08:
> will i be able to send list of markers to other users via waypoint ?

yes. same as other message types.

> Do I need to fix the firmware of the device and proto for this?

no. just the app side.

MaximKrasikov, 2022-09-09:
I apologize in advance for the stupid question. I just started to deal with meshtastic
I made changes to NodeInfo like EnvironmentMetrics in your last commits and add handleReceivedWaypoint in MeshService. Also create my custom marker in MapFragment via long click on osm view and save it in NodeInfo my new Waypoint, but nothing is sent. 
Are these all the changes I need?
my device-firmware 1.3.39 and 1.3.40 application in android
Thanks for adding osm)

andrekir, 2022-09-09:
feel free to ask questions. make sure firmware and app are on 1.3.40, there are lots of breaking changes with previous versions.

waypoints are handled as DataPacket instead of NodeInfo. quick draft:

- add WAYPOINT_APP_VALUE to recentDataPackets in MeshService (same as TEXT_MESSAGE_APP_VALUE)
- add waypoint object in model/MessageState and filters by type == TEXT_MESSAGE_APP_VALUE / WAYPOINT_APP_VALUE for each list
- from MapFragment use MessageState.waypoints.observe to build markers
- add logic to send waypoint messages similar to MessageState.sendMessage (initially just send to the primary channel).

also need to check how the waypoint payload looks like and how to get the data from DataPacket.bytes

MaximKrasikov, 2022-09-09:
Do I need to change something in MainActivity?
Need to make edits only in MeshService, MessageState, MapFragment and NodeInfo?

andrekir, 2022-09-10:
no changes in MainActivity or Nodeinfo. I added the DataPacket payload parser for Waypoints [here](https://github.com/meshtastic/Meshtastic-Android/blob/9c6a42e5c2e00584f30507f8e877462178f9beff/app/src/main/java/com/geeksville/mesh/DataPacket.kt#L58-L62). MeshService, MessageState and MapFragment should work. 👍

MaximKrasikov, 2022-09-12:
please add example in next commit. One may without using on map. Just functional, like your parser

andrekir, 2022-09-13:
check: https://github.com/andrekir/Meshtastic-Android/tree/waypoint-test

you can send a waypoint from the menu and it will show up on MapFragment.

MaximKrasikov, 2022-09-14:
thank you very much!

MaximKrasikov, 2022-09-15:
test point is initialized latitude=0 and longitude=0

andrekir, 2022-09-16:
@MaximKrasikov we're moving our packet database to Room, I added a waypoints livedata source in MapFragment you can use [here](https://github.com/meshtastic/Meshtastic-Android/blob/2b8b10a8cf811ec3a23afbfdb42286f1e8377c48/app/src/main/java/com/geeksville/mesh/ui/MapFragment.kt#L86-L89).

andrekir, 2023-02-01:
basic functionality has been added in #578 but still a few TODOs:
- [ ] waypoint editing
- [ ] display bearing/distance from local node
- [ ] add keyboard emoji picker
- [ ] add Expire date/time picker

prokrypt, 2023-05-08:
Would be nice if there were notifications for new/edited waypoints too~

yo3hjv, 2024-01-24:
Is there any way to add waypoints as a list of Lat/Lon/Name/Description/Icon from a file? It will be usefull in case of an emergency where a list of usefull POI can be activated at once.
The file can be prepared in time and actualised without cluttering the map when there is no need. When the emergency is declared, the file can be imported and the POI then sent to the group.
This can be usefull in emergencies; our emcomm team is set to go to some objectives to cover with radio communications some institutions. Thanks, YO3HJV

closed 2024-02-07 by andrekir