meshtastic/Meshtastic-Android #479: Provide Location To Mesh Checkbox Unchecks

ghostop14, 2022-09-08:
Running android version 1.3.40 on a galaxy s10, when a connection to a heltec v2.1 radio is established, "Provide location to mesh" unchecks after having been checked.

andrekir, 2022-09-08:
the box will uncheck when location service isn't running. usually permissions are missing or gps location provider is disabled.

ghostop14, 2022-09-08:
The GPS didn't have a lock but all location service permissions were allowed, so no-lock must've caused it.  I think the behavior I'd expect is that it would stay checked, just not provide coord to the mesh until they become available.  Like if you went in a building that had bad GPS reception, then came back out.  I'd expect coord to recover when lock was reacquired and start sending them again.

andrekir, 2022-09-08:
did you get it to work? no sat lock is fine. checked box indicates feature is enabled and working, and unchecked only if user action is required.

ghostop14, 2022-09-09:
I’ll be offline for a few days and do some more testing when I’m back.  I want to see if I can confirm if its just gps lock related.

ghostop14, 2022-09-15:
I do have a testing update.  So it seems when the GPS does not have a lock, it can automatically uncheck.  However, if it was originally checked, it does not recheck itself when GPS lock is reacquired.  Maybe an alternative UI approach would be the checkbox to indicate if position should be provided if available, and an indicator of GPS lock status / providing location to mesh rather than using the checkbox?

andrekir, 2022-09-21:
this was improved in 1.3.41, if there are any errors please reopen.

closed 2022-09-21 by andrekir