meshtastic/Meshtastic-Android #487: Unable to connect T-beam 1 with App

Discharge87, 2022-09-30:
Hello,

i cant connect to both of my devices with the android app on Galaxy S9 + S20plus. 
Binding phones with devices including the bluetooth code is working fine.

I have installed device firmware 1.3.41 and Android Appversion 1.2.67 and 1.2.64.

I also have an iphone 13 pro. The app from there is working nice.

Regards

andrekir, 2022-09-30:
you can't mix 1.3 and 1.2 versions, if you're using 1.3 firmware you need the 1.3 app.

https://meshtastic.org/docs/software/other/1.3-enthusiast#meshtastic-13---android

closed 2022-09-30 by andrekir

Discharge87, 2022-09-30:
> you can't mix 1.3 and 1.2 versions, if you're using 1.3 firmware you need the 1.3 app.
> 
> 
> 
> https://meshtastic.org/docs/software/other/1.3-enthusiast#meshtastic-13---android
> 
> 
> 
> 

Still not working with 1.3.41 on Android phone.

sachaw, 2022-09-30:
@Discharge87 we need a little more to go off than "it's not working" as mentioned above, please ensure your app and firmware versions match. If you are still having trouble please walk us through exactly what you do and what happens, also what you expect would happen.

Discharge87, 2022-10-01:
> @Discharge87 we need a little more to go off than "it's not working" as mentioned above, please ensure your app and firmware versions match. If you are still having trouble please walk us through exactly what you do and what happens, also what you expect would happen.

Hey,

i have installed the same 1.3.41 versions on all devices. (tbeams, android and iphone versions)
I said "its not working" because its the same error as before.

When i go in the Android app to the nodes- it doesnt show me them.  But i can couple the devices via bluetooth codes(6 numbers). So the system recorgnizes them but not the app. 
On the iphone the app its working fine since few days with both of my meshtastic devices. 

Regards

Discharge87, 2022-10-02:
[image]

garthvh, 2022-10-02:
Have you clicked the circular + icon in the bottom right and added your device?

Discharge87, 2022-10-02:
> Have you clicked the circular + icon in the bottom right and added your device?

Yes. 

And i tried to pair my phone and device before i started the meshtastic app and vice versa- no results. 

I also reinstalled the app and the firmware- also no results. (Same versions everywhere 1.3.41)

With the iphone app its working fine.

zeitrechnen, 2022-10-03:
Hi,

I think I have the same/a similar problem. On my Android devices with app version 1.2.67 and T-Beams with 1.2.65 I had no problem with pairing the devices. I upgraded the T-Beams yesterday to 1.3.41 and the apps also to 1.3.41 but in some cases I cannot successfully finish the pairing process:

- The phone discovers in the system menu the T-Beam and I can pair them. But when I try to add them inside the app, the modal window opens and searches for devices but then closes after a few seconds without showing any device.
- It also happens when I deleted the T-Beam in the bluetooth system settings before and try to start the pairing inside the app without the devices being paired.
- There a differences between the phones/OS versions: On a Huawei phone with Android 10 it works at least with the APK version from Github. On my Pixel 4a with GrapheneOS and Android 13 I have the issue with the version from the Aurora store, F-Droid and the Github APK. On the same device with Graphene and Android 13 there war no problem at all with the combination of the 1.2.67 app version and the T-Beam with 1.2.65.
- Connection with the iOS App is working fine.

Many thanks,
Christian

Discharge87, 2022-10-03:
@zeitrechnen 

Hey,

yeah, it looks like we have the same problem.

- Pairing works
- Connecting in app works not(it    doesnt show me the (paired) device there)

zeitrechnen, 2022-10-08:
@andrekir @sachaw Do you need more information? Any ideas what I can do to solve this issue?

andrekir, 2022-10-08:
@zeitrechnen please create your own issue or if you have questions post at https://meshtastic.discourse.group or join our Discord https://discord.gg/TYHeJpPc