meshtastic/Meshtastic-Android #491: Cannot pair T-Beam with app after upgrade to 1.3.41

zeitrechnen, 2022-10-08:
Hi,

On my Android devices with app version 1.2.67 and T-Beams with 1.2.65 I had no problem with pairing the devices. I upgraded the T-Beams yesterday to 1.3.41 and the apps also to 1.3.41 but in some cases I cannot successfully finish the pairing process:

- The phone discovers in the system menu the T-Beam and I can pair them. But when I try to add them inside the app, the modal window opens and searches for devices but then closes after a few seconds without showing any device. It also happens when I deleted the T-Beam in the bluetooth system settings before and try to start the pairing inside the app without the devices being paired.
- There a differences between the phones/OS versions: On a Huawei phone with Android 10 it works at least with the APK version from Github. On my Pixel 4a with GrapheneOS and Android 13 I have the issue with the version from the Aurora store, F-Droid and the Github APK. On the same device with Graphene and Android 13 there war no problem at all with the combination of the 1.2.67 app version and the T-Beam with 1.2.65.
- Connection with the iOS App is working fine.

Many thanks,
Christian

caveman99, 2022-10-08:
up to android 12 this should work: 
- 'forget' the device from the bluetooth system menu
- make sure to have the 'location' permission enabled for the meshtastic app
- search for new devices from the meshtastic app (plus sign in the lower right corner)

For Android 13 we're looking for more bug reports...

garthvh, 2022-10-08:
Make sure you forget any device pairings that are stored on your device in the system menu.

For android < 12 some phones require that location permissions be enabled for bluetooth to work this is a result of a security breach where BLE can be used to precisely locate a user.

The new 1.3 stack is actually 100% BLE now instead of the previous BLE / Classic hybrid, so you do not want to pair the devices in the system menu, only the app.

andrekir, 2022-10-09:
@caveman99 @garthvh no need to recommend location permission. if a permission is needed we ask for it and no location permissions are required for [Bluetooth API used](https://developer.android.com/guide/topics/connectivity/companion-device-pairing) on Android 8+ (95% our user base). some OS bug needs location turned on, not permission.

@zeitrechnen there are no changes in the app's Bluetooth since before 1.2.67 and there are tons of problems being reported for Android 13 we can't fix. the window showing Bluetooth devices is provided by the system, though I'm sure that is working on Pixel 4a stock and Graphene OS. try turning on location before scanning.

zeitrechnen, 2022-10-13:
Indeed, with location services turned on before starting the pairing process it works. Many thanks for your support!

closed 2022-10-13 by zeitrechnen