meshtastic/Meshtastic-Android #502: Feature Request: Add ability to permanently keep screen on

lesykm, 2022-10-22:
Sometimes I just want to watch for a messages in channel/nodes and not touch the phone but screen on the phone keeps turns off.
It's OK to keep this behavior by default but having checkbox in settings to keep screen on would be useful.

caveman99, 2022-10-22:
it's not in the App, but there's a config item. display.screen_on_secs - you can set it to a random high number, unsetting it means a default of 1 minute.

caveman99, 2022-10-22:
e.g. 3600 would be one hour

lesykm, 2022-10-25:
Sorry for confusion but I meant Android phone screen and not node OLED screen. Some apps like navigation, etc keep screen on  in order for user to keep watching it.

Solution should be similar to this one: https://www.stechies.com/keep-screen-stay-awake-android-app/

andrekir, 2022-11-03:
few reservations about this. there is no elegant way of changing window flags without an app restart, and there are known bugs and limitations for this to work.

for anyone interested adding this feature, make sure these are addressed before submitting a PR.

a third party app for this might be a better choice.

closed 2022-11-09 by andrekir