meshtastic/Meshtastic-Android #504: Occasional random notifications that app is disconnected on Android 12

Nestpebble, 2022-10-26:
Latest beta app (1.3.45) installed on Android 12 smartphone. Occasionally, 1-3 times per day, a random notification will appear that says Meshtastic is disconnected. In the app, it is set to connect to none(disable).
Surely the notification is erroneous?

PWRxPSYCHO, 2022-10-26:
Android 12, app version 1.3.45

pulled this info from the discord

Nestpebble, 2022-11-22:
Here's some relevant screenshots:

[image]
[image]

PR #551 (2023-01-03): update foreground service start
- update service `startForeground()` method

fixes #504
closed 2023-01-03 by andrekir