meshtastic/Meshtastic-Android #511: Feature Request: Disable notifications, and allow for direct messages.

hatesyourface, 2022-11-01:
I feel like soon I'll want a way to disable notifications for the default channel, but would likely still need them for direct messages. Has anyone already talked about this? Thanks

SenorFusion, 2024-04-17:
Any chance this could be revisited?

The issue is called "Mute Contacts", but I am asking about Muting Channels.

As meshes grow and default LongFast gets busier, this would be a nice quality of life improvement to be able to Mute specific channels.

Many people talk on default LongFast all day everyday, and I don't want to completely remove it and lose the publish mesh, but it would be really nice not to get a notification for every single one of their messages.

Thanks!

thatSFguy, 2024-04-20:
I think I'll submit a feature request under the firmware.  With the rise of T-Deck and other messengers, the channel mute should probably reside in the firmware instead.

PR #1003 (2024-04-28): feat: add `ActionMenu` option to mute contacts
closes #511

https://github.com/meshtastic/Meshtastic-Android/assets/48680741/167e624f-3f21-4844-afe6-2cfa316eba6f
closed 2024-04-28 by andrekir