meshtastic/Meshtastic-Android #527: Feature Request: Exit option that quits Meshtastic Android so it doesn't start back up until launched

prurigro, 2022-11-28:
Hey, I want to start by saying thanks to everyone involved in making this great piece of software!

So my issue is that I'm typically not connected to a meshtastic device, and the app stays closed if I never opened it. Once I do open it up and connect to a device, a permanent notification appears showing connected device as expected, and if I then go in and select "None (disable)" in the device selection screen and swipe away the app in the recents list the notification will be gone, which makes sense. After this however, I've been noticing the permanent notification eventually reappears with a message saying there are 0 connected devices, and I have to go back into the app and close it to make the notification go away again. This repeats until I force quit the app or reboot my phone.

I realize this could be a bug, but since "None (disable)" is also selected when there are no available devices and it might be handy to have the option to keep things running if a device reconnects, maybe the permanent notification could stick around by default once the app has been opened regardless of whether any devices are currently connected, and instead there could be an exit option in the menu that kills the app and prevents any hooks that might cause it to start back up again? That's how gadgetbridge works, and it works really well there (gadgetbridge also has an option to start at boot that might be worth borrowing too?).

Thanks for your time :)

andrekir, 2023-01-10:
thanks for the detailed description. exit buttons have been discussed before, check #186 
the root cause of the notification bug was tracked in #504 and should be resolved with the fix first released in 2.0.9.

closed 2023-01-10 by andrekir

prurigro, 2023-02-07:
That solution works too- thank you!