meshtastic/Meshtastic-Android #539: Meshtastic App can't see or connect to boards.

mcayea95, 2022-12-12:
Phone: Galaxy A53 running stock Android 13
Meshtastic App: 2.0.6.97
Tbeam Firmware: 2.0.6

I am having difficulty getting the app to recognize my tbeams. I can see and pair to the devices in the phone Bluetooth settings but when I go to add the device in the app, the devices do not show on the nearby device list.

I verified the Bluetooth connection works via web app client.meshtastic.org on my PC with no issues.

Any guidance would be greatly appreciated!

troca-hack, 2022-12-13:
same Issues
after several flashes several times and for a while I managed to connect, but since then nothing

mcayea95, 2022-12-13:
> Try 2.0.7 it has improved BLE connection options

I'll give it a shot!

mcayea95, 2022-12-13:
ALCON,

After an updater from 2.0.6 to 2.0.7, I can now see and pair with my boards. Thank you @garthvh for the response and solution!

closed 2022-12-14 by mcayea95