meshtastic/Meshtastic-Android #541: T-echo wont report back a temperature when its below 0°C

problex, 2022-12-14:
Hi,
Having a problem where the T-echo wont report back a temperature when its below 0°C in the Android app.
[image]

Here are the serial logs.

02:49:57 4799 [Router] (Received from T-c): barometric_pressure=1033.558472, current=0.000000, gas_resistance=0.000000, relative_humidity=46.426758, temperature=-2.850000, voltage=0

As you can see it is receiving the temperature correctly with the negative symbol -2.850000

Link to Discourse [](https://meshtastic.discourse.group/t/eviromental-sensor-wont-report-temperatures-below-0-c/6871)

Snowybirb, 2022-12-18:
Having the same issue on my RAK4631. Once temperature goes to a zero or negative number it won't display in the app but is showing up fine in the serial logs

andrekir, 2023-01-03:
fixed in #532

closed 2023-01-03 by andrekir