meshtastic/Meshtastic-Android #543: Feature request: clicking in nodes gps coordinates should open in the included meshtastic app, not in external app.

megabyte2010, 2022-12-21:
When i click on the gps coordinates in the nodes list, the phone asks me to open google maps (or in other app that can handle coordinates). I think it should open in the meshastic app.

caveman99, 2022-12-21:
the meshtastic app is just a viewer and will show the node anyway. Routing this via intent allows you to navigate to the target or feed them to other apps.

Ferretdog, 2024-01-12:
> the meshtastic app is just a viewer and will show the node anyway. Routing this via intent allows you to navigate to the target or feed them to other apps.

It opens in google maps for me - there is no option to open it in another app, or to copy it to paste it somewhere. White it centers on the location, it does not pin it (so it's not really made for you to navigate to it) and it's not placed in the search bar of google maps to allow it to be copied out or navigated-to either.

jesseas, 2024-10-20:
Just adding my vote to making it easy to go from the built in map centered on a node. 

I agree its great to be able to get the location into another app, but most of the time I just want to see where a node is.