meshtastic/Meshtastic-Android #545: Height above ground information missing

SR7673, 2022-12-25:
Hi all, 
Currently it is not possible to get the information about the height above sea level (ASL), right? I just see the GPS coordinates, not the height ASL. This can be an important information. Using V2.0.7 at the moment.
Best wishes for 2023.

Ferretdog, 2024-01-12:
This would be great while I'm paramotoring or for balloon tracking.

daviscodesbugs, 2024-02-10:
If I'm not mistaken, this information does come through with location update, so the remaining work to be done is displaying this in the app, right?
Where would you expect this to show up? In the pop up over a node in the map view?

daviscodesbugs, 2024-02-25:
#826 wants it in the node list

andrekir, 2024-02-25:
> this information does come through with location update, so the remaining work to be done is displaying this in the app, right? Where would you expect this to show up? In the pop up over a node in the map view?

right, altitude is just not shown. main challenge is coord formats like UTM, MGRS, etc. leave no room in the node list card to add it.

daviscodesbugs, 2024-02-25:
I'm not super familiar with coordinate formats; I'll do some reading and see if I can mock up some potential layouts

daviscodesbugs, 2024-02-28:
Here's an idea:

To me, the actual text of the coordinates isn't inherently useful since I don't really have an idea of where that is. Distance and elevation are useful, and it's nice to know that we _have_ a location. So replace the coordinates with an icon and keep the same linking out to a map with the coordinates functionality. This frees up some room to display elevation. When the icon is long-pressed, toast the coordinates and copy the value to the clipboard. 

What do you think?

[image]

b8b8, 2024-02-29:
i think its really important to show the actual location coordinates. its good for screen shots and provides the raw data. I dont think this info should be dumbed down.  In fact i think clicking and holding the coordinates should copy them to the clipboard, single click open them in google maps as it is currently.

b8b8, 2024-02-29:
> right, altitude is just not shown. main challenge is coord formats like UTM, MGRS, etc. leave no room in the node list card to add it.
 there seems to be lots of room just to the right of the coordinates on my android device with MGRS and UTM. maybe just put in brackets  (1200m) (1200ft)

daviscodesbugs, 2024-02-29:
> i think its really important to show the actual location coordinates. its good for screen shots and provides the raw data. I dont think this info should be dumbed down.  In fact i think clicking and holding the coordinates should copy them to the clipboard (like you said), single click open them in google maps as it is currently. these numbers would be also important if you needed to call out the location over a voice channel such as GMRS. Always provide more data. LAT/LONG are pretty fundamental to GPS based use cases (which meshtastic is).
> 
> 

Oh for sure. Not wanting to dumb it down or remove the information, just simplify the information shown in the node list. The exchange being a tap to open a "more info" page, for example

daviscodesbugs, 2024-02-29:
> > right, altitude is just not shown. main challenge is coord formats like UTM, MGRS, etc. leave no room in the node list card to add it.
> 
> there seems to be lots of room just to the right of the coordinates on my android device with MGRS and UTM. maybe just put in brackets  (1200m) (1200ft). make ASL a smaller font
> 
> what phone resolution and size did you use for those screenshot mockups? i have wayyyy more horizontal space on my pixel 7a.
> 
> [image]
> 

I wish it were that simple, but there are lots of things at play. For starters, you can change your font settings on your device and that affects the layout in app

b8b8, 2024-03-01:
my original suggestion for -More info on node page #826- was to add a single extra line to each of the node cards. this should solve the issues presented so far. We still have to scroll to see anymore than about 5 nodes so it doesnt really affect usability at all to do this

closed 2024-03-07 by andrekir