meshtastic/Meshtastic-Android #548: message order

seydunatirej, 2022-12-29:
When a message is sent repeatedly, the cloud shows as not received even though the message has reached the top. Aren't the messages in the queue one by one? I think they're all included at once

andrekir, 2022-12-29:
not an app issue. grab device logs to see what is going on (likely a `MAX_RETRANSMISSION` error). also a crossed cloud doesn't mean not received.

seydunatirej, 2022-12-30:
where exactly can i look

andrekir, 2022-12-30:
for device logs use any serial monitor program while connected to USB, or if you have the python CLI installed run `meshtastic --noproto`

if you need help check https://meshtastic.discourse.group or Discord https://discord.gg/TYHeJpPc

seydunatirej, 2022-12-30:
Thank you very much!

closed 2023-01-27 by andrekir