meshtastic/Meshtastic-Android #550: Feature Request: Add position history to the map view

pmarches, 2022-12-31:
Seeing a history of the last few positions for each node would be very nice. Having a few history points allows you to estimate heading and speed.

somnium78, 2023-01-17:
+1

i was just about to write a similar request. can the app use T-Beams rage test csv to show the last 10 or 20 positions?

ryanjjennings, 2023-01-23:
This would be a nice feature.  Maybe pick last x minutes of history to show?

mchuzie, 2023-02-04:
Absolutely. This would be a fantastic feature.

LO77ARO, 2023-02-14:
+1   this would be nice!

niksimms, 2023-02-20:
if this can trace for the last 10 entries that would be a good indication where it is going.

HikerScott, 2023-08-10:
Last 5-10 positions, or even hour, is waaay too short for my use case.

Preferably each node could be assigned a colour and the full pin and label would only be shown on the latest entry.  All other "breadcrumbs" would just be visible squares in the color of the node.

Ideally we'd be able to track 48 hours or more of breadcrumbs and have a button to clear the map of breadcrumbs.  If it was "X number of positions" with a configurable number I would want the max number to be very very large.

closed 2024-11-06 by andrekir