meshtastic/Meshtastic-Android #559: [Feature request] Features sync between Android an iOS apps

IZ1IVA, 2023-01-11:
I don't own an iOS 16 device, but I've discovered by chance in a [post](https://meshtastic.discourse.group/t/t-beam-and-device-metrics-log-only-storing-4-values/7044?u=iz1iva) a "Device Metrics Log" and graph tab, which is not present in the Android app.
Would it be possibile to add it to the Android app?
How about "syncing" features between the apps?

Cheers

geeksville, 2023-01-11:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/t-beam-and-device-metrics-log-only-storing-4-values/7044/9

hitachi8, 2023-10-01:
hello, if i might please bring my comment into this.

i would also love to be able to log, save and backup the telemetry data from a node of my choosing unto my own smartphone in order to make a wall display, or graph later on. basically have the data available on that cellphone.
here would be my setup:
[distant WX station] --LoRa--> [personal node] --> store received telemetry data 

also if possible: 
-save it automatically instead of having to do it manually?
-have the data in a .txt file line by line format as shown in the IOS app (easier to use)
-being able to select what node to store said telemetry data from
-input where to save the .txt file 

Thank you all for your work.

closed 2024-08-31 by andrekir

IZ1IVA, 2024-09-02:
Cheers!