meshtastic/Meshtastic-Android #601: [UI BUG] Can not set region Pixel 4a

ramaboo, 2023-03-09:
On a Pixel 4a you can't set a region. 

See these screenshot. No matter where you click on the select drop down it won't set the region. It just stays unset.

[image]
[image]

andrekir, 2023-03-09:
what device and firmware version?

ramaboo, 2023-03-09:
Pixel 4a Android 13. Mestastic android app 2.0.22.

andrekir, 2023-03-09:
device can be Tbeam/RAK/etc and the Meshtastic software version it is running.

andrekir, 2023-03-10:
apps can't communicate with devices running old firmware. looks like this is your issue. try flashing the latest 2.1.x: https://meshtastic.org/docs/getting-started/flashing-firmware 

if that doesn't work let me know.

closed 2023-03-10 by andrekir