meshtastic/Meshtastic-Android #602: fdroid publishing

fds2610, 2023-03-11:
AS you habe a fdroid release, i assume you plan to publish meshtastic Tiere but currently i cannot See it in fdroid. Do i need to customize repositories?

caveman99, 2023-03-11:
It's in the process of being added to fdroid proper. Right now it lives in its own repository that will be sunset after the main fdroid repo carries it.

fds2610, 2023-03-12:
Hi caveman, THX for ur quick answer. Do u know a timeframe when its available?

andrekir, 2023-03-12:
while we wait for inclusion follow the docs: https://meshtastic.org/docs/software/android/installation#install-with-f-droid

closed 2023-03-12 by andrekir

licaon-kter, 2023-11-21:
Hmm, recipe:
```
  - versionName: 2.2.13
    versionCode: 30213
    commit: 4be5648746263f2f8a1fb25ba644a216f025fb33
    subdir: app
    submodules: true
    sudo:
      - apt-get update
      - apt-get install -y openjdk-17-jdk-headless
      - update-java-alternatives -a
    gradle:
      - fdroid
    rm:
      - app/src/main/assets/firmware
      - app/src/main/res/values/curfirmwareversion.xml
    prebuild:
      - sed -i -e '/\"useCrashlytics\"/,/^        }/d' ../build.gradle
      - git clone https://github.com/meshtastic/firmware.git
      - pushd firmware
      - export CUR_FIRMWARE_VERSION=$(git tag --sort version:refname|tail -1)
      - export SHORT_FIRMWARE_VERSION=$(echo $CUR_FIRMWARE_VERSION|sed -e 's/^v//'
        -e 's/.\{8\}$//')
      - popd
      - rm -fr firmware
      - echo -e "<resources>\n    <string name=\"cur_firmware_version\" translatable=\"false\">$CUR_FIRMWARE_VERSION</string>\n    <string
        name=\"short_firmware_version\" translatable=\"false\">$SHORT_FIRMWARE_VERSION</string>\n</resources>"
        > src/main/res/values/curfirmwareversion.xml
      - mkdir -p src/main/assets/firmware
```

can somebody test the APK? 
[com.geeksville.mesh_30213.apk.ZIP](https://github.com/meshtastic/Meshtastic-Android/files/13429449/com.geeksville.mesh_30213.apk.ZIP) _(remove .ZIP)_

andrekir, 2023-11-21:
@licaon-kter can we use reproducible builds? we publish F-Droid APKs on all [releases](https://github.com/meshtastic/Meshtastic-Android/releases) (ie. [fdroidRelease-2.2.13.apk](https://github.com/meshtastic/Meshtastic-Android/releases/download/2.2.13/fdroidRelease-2.2.13.apk))

thank you!

licaon-kter, 2023-11-21:
Dunno, that firmware version stuff needs to be a static here, not something that might vary between yesterday when you've built it and tomorrow when I build it.pp

andrekir, 2023-11-22:
@licaon-kter would it be helpful if we remove the firmware assets logic along with `curfirmwareversion.xml` (it is unused at the moment)? if so I can work on this for the next release.

also, there are Crashlytics dependencies in both project (root) and /app `build.gradle` files. using the commands below is guaranteed to remove them and also future proof:

```
sed -i -e '/com.google.gms/d' -e '/firebase-crashlytics-gradle/d' ../build.gradle
sed -i -e '/com.google.gms/d' -e '/com.google.firebase.crashlytics/d' build.gradle
```

licaon-kter, 2023-11-22:
> would it be helpful if we remove the firmware assets logic along with curfirmwareversion.xml (it is unused at the moment)? 

yes @andrekir ping us when ready to retest

Coehill, 2024-03-21:
@andrekir Should this ticket be reopened?

licaon-kter, 2024-03-22:
```
Categories:
  - Connectivity
  - Internet
  - System
License: GPL-3.0-only
WebSite: https://meshtastic.org
SourceCode: https://github.com/meshtastic/Meshtastic-Android
IssueTracker: https://github.com/meshtastic/Meshtastic-Android/issues
Changelog: https://github.com/meshtastic/Meshtastic-Android/releases

AutoName: Meshtastic

RepoType: git
Repo: https://github.com/meshtastic/Meshtastic-Android.git
Binaries: https://github.com/meshtastic/Meshtastic-Android/releases/download/%v/fdroidRelease-%v.apk

Builds:
  - versionName: 2.3.1
    versionCode: 30301
    commit: be741a9200f302404240ccb8db88b36cd603f513
    subdir: app
    submodules: true
    sudo:
      - apt-get update
      - apt-get install -y openjdk-17-jdk-headless
      - update-java-alternatives -a
    gradle:
      - fdroid
    prebuild: sed -i -e '/(useCrashlytics)/,+3d' -e '/firebase/d' -e '/gms/d' ../build.gradle
      build.gradle

AllowedAPKSigningKeys: a93b456568c175db0800a09f06777f892d812432adb8a3df73bc3e7f06c80c6d

AutoUpdateMode: Version
UpdateCheckMode: Tags
CurrentVersion: 2.3.1
CurrentVersionCode: 30301
```

builds fine and repro :tada: 

Can anyone test the APK? 
[com.geeksville.mesh_30301.apk.ZIP](https://github.com/meshtastic/Meshtastic-Android/files/14721833/com.geeksville.mesh_30301.apk.ZIP) _(remove `.ZIP` from name)_

andrekir, 2024-03-22:
@Coehill please follow the progress in #6 or the fdroid gitlab request for packaging (below).

@licaon-kter unfortunately 2.3.1 is not good for production. our last stable release is still 2.2.23 https://github.com/meshtastic/Meshtastic-Android/releases/latest

we're trying to land a stable 2.3 release, as soon as we have one I will follow up in https://gitlab.com/fdroid/rfp/-/issues/2340

hopefully soon! thanks for checking on us!

licaon-kter, 2024-03-22:
ping when ready :+1:

Coehill, 2024-03-22:
@licaon-kter I was able to download it successfully, connect to a node, send a message, and change some settings.