meshtastic/Meshtastic-Android #621: Android app vs 2.1.4 , peers show distance in km in stead of miles

RightHandMan1, 2023-04-16:
Version 2.1.4 of the android, (2.1.17 firmware), in device settings I have Imperial selected, but the peers list keeps showing their distance in km. this node IS connected to wifi and mqtt, but assume it should still translate the distance.

caveman99, 2023-04-16:
device settings only alter the on-display appearance for the device, not the representation in the app. I think the app gets the unit settings from the phone locale.

RightHandMan1, 2023-05-24:
seems not, the phone is set to all-merican , still show peers in km, node display shows in miles properly. Still this way in ver 2.1.10

RightHandMan1, 2023-06-07:
Still this way in 2.1.14 android app. Since the app can change the node, read the node, it should follow the node settings.

PR #651 (2023-07-06): Add option for Miles and Feet for distance
closes #621
closed 2023-07-06 by andrekir