meshtastic/Meshtastic-Android #625: Fixed Position - Bug

semtex1987, 2023-04-20:
When attempting to set a fixed position for a node via the Android app, attempting to type a - in the longitude box does not work, the combined .- button does not allow you to long press to type a - and does not type a . either.

Copying and pasting longitude coordinates into the longitude box which includes a - to bypass this issue does not allow you to save the information either, perhaps a normalization issue with this field?

tim292stro, 2023-04-20:
A node with uBlox GPS should probably also use the provided "survey-in" feature to stabilize the position, changinge the filtering mode from moving to static.

garthvh, 2023-04-20:
> A node with uBlox GPS should probably also use the provided "survey-in" feature to stabilize the position, and changinge the filtering mode from moving to static.

We use NMEA for GPS, there are no UBLOX specific features anymore.

andrekir, 2023-04-24:
@semtex1987 can you detail how you are typing minus? looks like it can't be typed when the field is empty, only in front of a number. copy/pasting works for me though.

semtex1987, 2023-04-26:
> @semtex1987 can you detail how you are typing minus? looks like it can't be typed when the field is empty, only in front of a number. copy/pasting works for me though.

Sure

Flip the switch to enable "use fixed position" which reveals the latitude and longitude fields.

They default to 0.0 for both.

Type in any coordinates in longitude, move cursor all the way to the left and then long press the .- button just makes the keyboard underline the first character to the right of where my cursor is. Short press does the same thing.

If I copy and paste into the longitude box coordinates with the - already there, the app then won't allow me to save the coordinates.

App version is 2.1.9

andrekir, 2023-04-26:
> App version is 2.1.9 on a Samsung Galaxy S23 Ultra

using Samsung Keyboard?

semtex1987, 2023-04-26:
> > App version is 2.1.9 on a Samsung Galaxy S23 Ultra
> 
> using Samsung Keyboard?

Yep, that was the issue. Switched to gboard keyboard and now I can type a - and save in the longitude field.

Apologies for wasting your time.

andrekir, 2023-04-26:
no worries, glad we found the cause. the Samsung Keyboard is using some strange logic for the `.- `key, where it's not short/long press, instead it outputs a dot first then on second press replaces the dot with a minus.

on the topic, I'd stay away from Samsung Keyboard or any keyboard with full network access. OpenBoard is a popular alternative.

either way, I'll add that logic to the validation so it also works correctly for Samsung users.

commit c821eb36 (2023-04-26): fix #625: handle Samsung Keyboard dot-minus key in TextField validation

Samsung Keyboard numerical keypad features a combined '.-' key that outputs a dot (.) on first press and replaces it with a minus (-) on second press. there is no option to output each symbol separately (short or long press, etc).

updated validation logic to handle dot symbol at the start of the input string.
closed 2023-04-26 by andrekir