meshtastic/Meshtastic-Android #635: feature request: let us select which channels to generate URL/QR for

prokrypt, 2023-05-16:
maybe have primary channel be required, but let us choose which secondary channels to share in case we don't want someone in our admin channel...

Ferretdog, 2024-01-12:
This, but also the ability to import just that channel - and not to replace your entire channel set with the one you are importing. 

I want to share a channel with friends without them having to manually add it (requiring me to send the encryption key using a clunky cut and paste), or remove all their own channels to accept it. It would ideally check that the modem preset and channel were the same before importing.

SenorFusion, 2024-04-25:
> This, but also the ability to import just that channel - and not to replace your entire channel set with the one you are importing.

This is the really important part.

Now that this is available on iOS and there is a working method, would it be easier to implement for Android?

This is a real hold-up for on-boarding other families and groups who are not tech savvy enough to use the CLI or web flasher for channel management

closed 2024-07-02 by andrekir