meshtastic/Meshtastic-Android #642: [Feature request]: GPIO switchable via app

SR7673, 2023-06-05:
The recent improvements to the app are just great!
It would be great to add the ability to switch GPIO pins via the app.
Just like https://meshtastic.org/docs/hardware/peripheral/ but not via CLI but with the app. 
Also it is / would be possible to request the current state (on/off) of a GPIO pin.

SR7673, 2023-06-05:
Hm, well... Assuming the gpio channel has been created: A sub-group "GPIO" (or slider or how you want to call it) could be shown, just like "LoRa" or "Device" or "Bluetooth". On the page of this sub-group, the available GPIO Pins (or let's say 3 commonly available) could be requested. Their state could be shown as green (On) or grey (Off), similar to other switches). By default they should be treated as Inputs, with the option to change them into outputs and to switch them On/Off.

pdxlocations, 2023-06-07:
I like the idea of "quick chat messages" on the gpio channel. "Lamp on" sends equivalent of "gpio pin 4 on" for example. may need to configure variables "name", "pin", "action", "destination".  You could leave off "destination" if its possible to DM to a gpio channel?

andrekir, 2023-06-08:
that doc link is ancient so take it with caution. there is a new "Remote Hardware" module (which is already in Radio Configuration). AFAIK the firmware feature is still a work in progress (not ready for use) but its all available for testing since 2.1.13.

for now it is too unclear if/what needs to be added for a UI.

@pdxlocations you are entering uncharted waters and would have to test. 😄

pdxlocations, 2023-06-08:
I'd be happy to test!  I have a node setup on my bench to experiment with GPIO and Remote Hardware.  I can turn an LED on and off remotely with the Python CLI.  I'll be keeping an eye on these features and will let you know if I come up with any brilliant UI ideas.

SR7673, 2023-06-11:
Well, i just tested:
Located IO7 on a RAK4631 (https://meshtastic.org/docs/configuration/rak-gpio-mapping) and connected a voltage probe to it. The GPIO channel has been created as well.
I can actually read 3.3 V when executing meshtastic --gpio-wrb 28 1 --dest '!bd6428c6'
Also this can be set to LOW / 0V (actually 0.00 V ) when executing meshtastic --gpio-wrb 28 0 --dest '!bd6428c6'
So this is the main functionality, which is great and most useful.
I can also confirm that it works over a remote node in the mesh, also most important.

Then i wanted to set this pin in Android version 2.1.14 using the Remote Hardware section. Adding this pin as type DIGITAL_WRITE is possible but the settings are not saved and after a reboot they are lost!

Probably this is a small bug which can be found quickly. So we are close to the desired functionality! Great! Looking forward to this...
Regards

andrekir, 2024-11-12:
`RemoteHardware` was removed from the firmware in https://github.com/meshtastic/firmware/pull/4890. closing.

closed 2024-11-12 by andrekir