meshtastic/Meshtastic-Android #643: [Feature Request]: Entering an IP address manually

G2Dolphin, 2023-06-05:
We have pretty active local Meshtastic community with couple of members having stationary placed DIY nodes, which are manageable through WiFi. Some members, me including, use VPN to manage those nodes, and the only option to do that from the phone application — to search for nodes being in the same network then keep application from being unloaded from RAM, so the address being kept on the nodes list.

Ability to manually specify IP address to connect would gladly help managing those nodes and seem to be a minor and not difficult update. :)

andrekir, 2023-06-05:
Duplicate of #472

closed 2023-06-05 by andrekir

G2Dolphin, 2023-06-05:
Sorry, should've searched for similar requests first!