meshtastic/Meshtastic-Android #649: [FEATURE] Custom connection options

tb0hdan, 2023-06-30:
Hi,

Please add the following:

- Custom IP:Port for Meshtastic devices behind router (and possibly HTTP basic or similar)
- Keepalive / Timeout / Reconnect options for WiFi + request above

Much appreciated.

ut3uku, 2023-07-01:
I consider it's important!

G2Dolphin, 2023-07-01:
Yeah, I already [https://github.com/meshtastic/Meshtastic-Android/issues/643](posted) similar FR, which is similar to [https://github.com/meshtastic/Meshtastic-Android/issues/472](other FR) before.

jp-bennett, 2023-07-17:
It doesn't sound like a good idea to expose a Meshtastic device to the Internet, even on a random port. A VPN like Wireguard is a much better approach. PR #663 does allow that use case.

PR #684 (2023-08-13): feat: implement TCP interface reconnections
Implement basic TCP interface reconnection attempts:

- Use an [exponential backoff strategy](https://en.wikipedia.org/wiki/Exponential_backoff): wait for increasingly longer intervals between reconnection attempts. Initial backoff delay (`MIN_BACKOFF_MILLIS`) is _1 second_, and doubles on each failure (_2^n_) to a max (`MAX_BACKOFF_MILLIS`) of _5 minutes_. No jitter is needed (for collision avoidance);
- Maximum reconnection attempts: a maximum number of times to try can be specified (`MAX_RETRIES_ALLOWED`), for now defaults to Int.MAX_VALUE.

close [... cut]
closed 2023-08-13 by andrekir