meshtastic/Meshtastic-Android #652: Build Fails

jp-bennett, 2023-07-06:
Android Studio can't find usb-serial-for-android:3.4.6. I've had to change that line to

`implementation 'com.github.mik3y:usb-serial-for-android:v3.4.6'`

at https://github.com/meshtastic/Meshtastic-Android/blob/05327d2c05bd40662c130b6caa61c1f09bcef7e4/app/build.gradle#L219

Not sure if this is just a problem on my machines. But it was required to compile successfully.

andrekir, 2023-07-06:
every code changes trigger CI builds using GitHub Actions, so we know when anything breaks https://github.com/meshtastic/Meshtastic-Android/actions

thats what the "checkmark" in front of commits represent (build success/failure). so definitely no build issues in the code.

you can also enable GitHub Actions on your fork's Settings.

jp-bennett, 2023-07-06:
It would be useful to understand this particular problem, to add a step to the build instructions. I've replicated this on two different machines, a Fedora and a Pop_OS machine.

andrekir, 2023-07-06:
odd. the standard format for library versions in Gradle has no prefix and the [lib repo](https://github.com/mik3y/usb-serial-for-android#quick-start) doesn't suggest that notation. you can see task ':app:compileFdroidDebugAidl' passed the latest [CI running Ubuntu 22.04.2 LTS](https://github.com/meshtastic/Meshtastic-Android/actions/runs/5479463850/jobs/9981520604#step:7:17).

andrekir, 2023-08-01:
closing. if anyone else is having issues please comment below.

closed 2023-08-01 by andrekir