meshtastic/Meshtastic-Android #657: [Bug]: Update Android app to correctly recognize NodeDB Lite changes

tropho23, 2023-07-09:
It appears the Android app is ignoring/not recognizing recent changes included in the new NodeDB Lite introduced in firmware 2.1.16.

The issue I noticed, and confirmed with much testing is GPS coordinates for each node are no longer included in the Nodes tab.  If the user manually sends a position update from a device (double press user button), the GPS coordinates will eventually show up for that node on the node list, but those coordinates will disappear after the next message is received from that node; the app no longer recognizes the position data included (related to mynodeinfo I believe).

Devices with firmware 2.1.15 work fine, as the new NodeDB Lite changes were implemented in 2.1.16.  I have been able to reproduce this with any firmware version after 2.1.15, and for all devices with GPS modules.

prokrypt, 2023-07-09:
Relevant: https://github.com/meshtastic/firmware/pull/2604

andrekir, 2023-08-01:
no update is needed

closed 2023-08-01 by andrekir