meshtastic/Meshtastic-Android #662: [Feature request] Blue dot for current location on map

ondrejtoral, 2023-07-16:
My own location updates are too infrequent for navigation. Could we get an indication of current position provided from phone gps in the map? Would a "blue dot" be difficult to implement? Thanks for considering it.

andrekir, 2023-07-17:
you can enable "provide location to mesh" for more precise location updates and disable when not needed to save battery. should achieve the same result. I like the idea of using the "blue dot" to differentiate our local node.

closed 2023-08-01 by andrekir

GPSFan, 2023-08-18:
This works fine when stationary, but when moving, OSM uses a white arrow to show the direction, which is also ok sometimes when the map is a satellite image. On a topo, the map is often mostly light colored or white and the white arrow melts into the background.
I added "setDirectionIcon(it.toBitmap())" in between the "setPersonIcon()" and "setPersonAnchor()" statements at about line 161 in MapFragment.kt
This makes it a blue dot all the time.
The "enableFollowLocation" at line 159 makes the map center on the phone's (tablet's) position which is a wonderful feature, but it should be noted that if you scroll or zoom the map, the folllowLocation feature is disabled by OSM. Re-enable by clicking the button off then on again.

PR #675 (2023-08-01): feat: add option to show phone GPS location on map
Adds OSM `MyLocationNewOverlay` to map for displaying current phone GPS location.

closes #662