meshtastic/Meshtastic-Android #727: [Bug]: OsmDroid: Problem downloading MapTile

prurigro, 2023-09-11:
I don't seem to be able to download maps at the moment. The full error is:

```
OsmDroid: Problem downloading MapTile: ##/####/##### HTTP response: Not Found
```

Thanks for your time!

andrekir, 2023-09-11:
for which map source?

andrekir, 2023-09-11:
osmdroid only gives us the error count, so there's not much to go off here. its all third-party and free map sources for offline tile downloads are few and limited. Open TOPO is also running under reduced capacity for the past couple weeks.

andrekir, 2023-09-12:
[refactor: change default map source to `Mapnik`](https://github.com/meshtastic/Meshtastic-Android/commit/4fa9c8f3cb9f7462487b5ab081a43830f289b3e4) 👍

new installs will default to the [MAPNIK](https://github.com/mapnik/mapnik) source with extensive coverage and better user experience.

closed 2023-09-12 by andrekir

prurigro, 2023-09-12:
Awesome, thanks so much!