meshtastic/Meshtastic-Android #766: Range test messages not showing up (only in debug)

marcluer, 2023-10-27:
Sony Xperia 10 II (Sender)-> Tbeam (sx127x) -> Tbeam (sx126x) -> Google Pixel 7 (Receiver)
Meshtastic App 2.2.12 on both Androids
Meshtastic Firmware tbeam 2.2.11.10265aa on both Tbeams

A few weeks ago I did my first range test and everything worked fine as expected:
Sender: "Range test" = enabled
Receiver: "seq 1" ... "seq 2" ... "seq 3" was showing up in the messages tab.

Fast forward to today and the range test messages can only be seen in the debug 
panel (receiver side), while manually typed messages go through and show up in the
messages tab without a problem.

I tested with different channels, upgraded both Tbeams to latest beta firmware,
turned the pixel to range test sender and the sony to receiver, but no luck.

Maybe this is simillar to [#209](https://github.com/meshtastic/Meshtastic-Android/issues/209) ?
But none of the mentioned workarounds worked for me.

Thanks!
Marc

garthvh, 2023-10-27:
Needs to be enabled on both devices, and no longer works on the public mqtt to manage spam messages.

marcluer, 2023-10-29:
Enabling range test on both devices works! 👍

I thought I previously only had to enable range test on one device...

closed 2023-10-29 by marcluer