meshtastic/Meshtastic-Android #774: [Feature Request] Node BLE connection priority

jeremiah-k, 2023-11-04:
It would be great if we could assign node priority values to each of our remembered nodes.

I have 3 main nodes that I connect to throughout the day and I have to manually switch to each one, depending on where I'm at.  For example with this feature we could set #1 to the Home node, so that's always preferred whenever it's in range, #2 to the Vehicle node, and #3 to the backpack node.

So when moving from the house to the vehicle and the house node becomes out of range, it will connect to the vehicle node, when leaving the vehicle it will connect to the backpack node, and so on.

When returning to the vehicle or to the house, it will connect to the higher priority nodes when detected (checking once a minute or however it's configured).

mattdrum, 2023-11-28:
I second this request. I have three primary nodes that I need to switch between during the day, (Home, vehicle, work). This feature would ensure that I stay connected when moving locations.

mtsdawson, 2024-02-04:
Same use case, I also have three nodes that would be set to preferred.

Expanding on the use case examples, I would apply this configuration to my wife's vehicle. In an emergency, it's unlikely that she would manually switch between nodes. This auto switching feature would remedy this problem and ensure she stays connected at all times.